nortd/driveboardapp #8: Serial connection unreliable

labdaemons, 2016-09-21:
Not establishing connection to machine.  Sometimes works, but rarely and unpredictable.

ckortenhorst, 2016-10-29:
I have same issue on and off running from command line and not init.d script seams to work much better.

stefanix, 2016-11-06:
I would need more info on this. Can you confirm this is actually about the serial connection between the BBB and the Atmega in a standard Lasersaur setup. Please describe how you have tested this and what errors you get. Any log output and platform info would also be interesting.

DSuds, 2017-02-08:
Just trying the switch to the driverboardapp from our working lasersaurapp/14.01 machine and we see a failure to connect pretty consistently.  So far it has worked 2 out of say 15 times.
It will sometimes flash the board successfully and always reports that reset succeeded.
Not sure how to debug the state of things.

stefanix, 2017-02-09:
Are you positive this is not the network/wifi connection being spotty? I have just posted an update that makes the status reporting more robust (http vs websocket). Also a common cause for problems is when the grounding to the lasersaur frame gets disconnected (typically laser PSU to frame).

Otherwise I think this may be a general ESD edge case. Moving around cables often solves this. If the problem persists let me know. Currently just finalizing the DriveboardUSB. It has opto-isolated sensor/control wires and a metal enclosure tied to AC ground. My suspicion is that some laser PSUs are noisier than others leading to some Lasersaur configuration being more difficult to get stable.

DSuds, 2017-02-09:
We use ethernet from the BBB and the laser power supply was not on (I have made the modification to control the motors and laser PSU with the eStop).  Not sure how spotty network would result in "Machine not found" from the backend but so be it.

I was logged into the BBB and used the old Lasurapp to flash the new firmware onto the arduino and things seemed better so it may be some sensitivity in the reset logic or an odd state the firmware was in.

But seems up and running at this moment so we will have some input on the new release in no time. :)

stefanix, 2018-05-19:
Using some cleaned tuned up serial code in v18.05. Should fix some edge cases of serial communication.

closed 2018-05-19 by stefanix