nortd/driveboardapp #21: Cannot connect serial on port: /dev/ttyACM0

magdesign, 2018-03-10:
I am trying since hours to upgrade my **LasersaurApp v.14.11b** to **DriveboardApp 17.04** beta.
After flashing the new firmware to my Arduino Mega 2560 (tried with: firmware.driveboard1403.hex and firmware.driveboardusb.hex and also with the original on the board).
I start the driveboardapp with: `sudo python driveboardapp/backend/app.py`

Setting the right serial port with: http://127.0.0.1:4444/config/serial_port//dev/ttyACM0

But I always get following error message:

```
ERROR: Cannot get 'hello' from controller
ERROR: Cannot connect serial on port: /dev/ttyACM0
WARN: Cannot connect to configured serial port.
INFO: Trying to find port.
ERROR: No controller found.
```
Also checked:
`/home/<myuser>/.driveboardapp/config.json`
 It looks like this:
```

{
    "network_port": 4444, 
    "feedrate": 2000, 
    "raster_leadin": 20, 
    "users": {
        "laser": "laser"
    }, 
    "seekrate": 6000, 
    "grid_mm": 100, 
    "firmware": "driveboardusb", 
    "serial_port": "/dev/ttyACM0", 
    "intensity": 0, 
    "max_jobs_in_list": 20, 
    "fill_leadin": 10, 
    "kerf": 0.3, 
    "workspace": [
        1220, 
        610, 
        0
    ], 
    "network_host": "", 
    "pxsize": 0.4
}
```

Is there something to change, e.g.  "firmware": "driveboardusb" ?
I tried on OSX, Ubuntu, and Windows. 
Any help very appreciated.

stefanix, 2018-03-10:
Try an Arduino Uno. I doubt the firmware is compatible with the Mega.

stefanix, 2018-03-10:
Happy to help when running on the correct hardware.

closed 2018-03-10 by stefanix