nortd/driveboardapp #28: Modifying and compiling firmware

schwittlick, 2019-03-24:
Hey everybody

I'm planning to modify the firmware a little, to support a third stepper in the z-axis.

Though I am having trouble compiling the firmware. There's a flash.py, that should do the job. Though it's trying to compile gcode.h/.c which is not in the project. When removing that, having other compilation errors.

I'm on Windows 10, using Arduino 1.6.7 and the driveboard app master branch from here.

Am I doing anything obviously wrong?

Thanks a lot for the help! Sending greetings :)

stefanix, 2019-03-24:
Hi Marcel,

There is a "rebuild firmware" in the burger menu of the UI. If you want to
know how it is done from python search for "flash" in the backend/web.py
and backend/driveboard.py files.

Z-axis support is working. Unless you need something fancy there are no
changes required. For example you can send a url request /movez/100 and it
will move z to 100.

Stefan Hechenberger, MFA, MPS
work: http://nortd.com
project: http://lasersaur.com
collab: Google AMI
instagram: @stefavon

schwittlick, 2019-03-25:
Hey @stefanix 

I tried that, the compilation fails, though:

[image]

This doesn't really tell me a lot about what's going on. And when attempting to compile manually, via flash.py this is happening:

[image]

And for the z-axis to work, I assumed I'd have to enable this flag in config.driveboardusb.h:
`// #define ENABLE_3AXES            // enable/disable 3-axis mode (vs 2-axis)`