nortd/driveboardapp #32: SyntaxError: invalid syntax on startup

ckortenhorst, 2023-12-10:
I am trying to upgrade from laserapp 14.11 using a BBB 

Did run following 
git clone https://github.com/nortd/driveboardapp.git
cd driveboardapp
scripts/install_packages.sh
scripts/upgrade_to_driveboardapp.sh
sync
reboot

```
Starting driveboardapp with --debug ...
Traceback (most recent call last):
  File "/root/driveboardapp/backend/app.py", line 40, in <module>
    import web
  File "/root/driveboardapp/backend/web.py", line 19, in <module>
    import driveboard
  File "/root/driveboardapp/backend/driveboard.py", line 1164
    print(f'WARN: pxsize of {pxsize_y} mm/px is too small. Setting to 0.01 mm/px')
                                                                                ^
SyntaxError: invalid syntax
```

Any thoughts?

IMG on BBB lasersaur-BBB-image-v15.01.img.xz
Running Python 2.7.6

martinxyz, 2023-12-12:
Not sure what the solution is, but that syntax is only valid from Python 3.6. Probably you'd have to upgrade the BBB base image/system. The line from the error message doesn't exist in the master branch, not sure which branch you're really using. May the branch's README.md will tell you what base system you need.

martinxyz, 2023-12-12:
I just rediscovered the instructions I wrote in my own fork (https://github.com/martinxyz/LasaurApp), so basically I had issues (bugs) with the Python3 version of that BBB image. So even if you can install Python3 it may be simply too old to run any recently written Python code.

(Btw. we still use the Pulseraster fork in our FabLab, but it's not maintained/updated any more. I'm not sure what the best fork would be to use these days. Pulserster still has a small number of additional bugfixes and the somewhat clunky raster GUI that kind-of-works, but isn't perfect or convenient. And you'd need to update the BBB image too, which is a bit annoying to do.)

scottshambaugh, 2024-02-26:
@ckortenhorst I would also flag my fork as a python3-compatible version with some more bugfixes and features: https://github.com/scottshambaugh/driveboardapp